hkust-smartcar/libsccc #20: Breakpoint triggered after sending buffer via bluetooth frequently

chunkitmax, 2015-01-21:
<ul>inno14-d-2014-inno.elf	
	<ul>Thread [1] <main> (Suspended : User Request)
		<ul>HardFaultHandlerC() at vectors.c:443 0xb7fa	</ul>
		<ul> signal handler called() at 0xfffffff9	</ul>
		<ul>_malloc_r() at 0x12808	</ul>
		<ul>operator new() at 0x12370	</ul>
		<ul>libsc::k60::UartDevice::SendBuffer() at uart_device.cpp:205 0x9872	</ul>
		<ul>main() at main.cpp:78 0x711a	</ul>
	</ul>
</ul>

<table>
<tr><td>stacked_r0</td>	<td>volatile unsigned long</td>	<td>536806660</td></tr>
<tr><td>stacked_r1</td>	<td>volatile unsigned long</td>	<td>536869764</td>	</tr>
<tr><td>stacked_r2</td>	<td>volatile unsigned long</td>	<td>4</td></tr>
<tr><td>stacked_r3</td>	<td>volatile unsigned long</td>	<td>536869884</td></tr>
<tr><td>stacked_r12</td>	<td>volatile unsigned long</td>	<td>0</td></tr>
<tr><td>stacked_lr</td>	<td>volatile unsigned long</td>	<td>74557</td></tr>
<tr><td>stacked_pc</td>	<td>volatile unsigned long</td>	<td>75798</td></tr>
<tr><td>stacked_psr</td>	<td>volatile unsigned long</td>	<td>16777216	</td></tr>
<tr><td>_CFSR</td>		<td>volatile unsigned long</td>	<td>1024</td></tr>
<tr><td>_HFSR</td>		<td>volatile unsigned long</td>	<td>1073741824</td></tr>
<tr><td>_DFSR</td>		<td>volatile unsigned long</td>	<td>1</td></tr>
<tr><td>_AFSR</td>		<td>volatile unsigned long</td>	<td>0</td></tr>
<tr><td>_BFAR</td>		<td>volatile unsigned long</td>	<td>3758157112</td></tr>
<tr><td>_MMAR</td>		<td>volatile unsigned long</td>	<td>3758157108</td></tr>
</table>

nkming2, 2015-01-23:
I could not reproduce the bug. I've been looping the SendBuffer() call for 5 minutes and nothing bad has happened. Please provide a tiny code snapshot that could reproduce the bug, thanks.

chunkitmax, 2015-01-27:
```C++
array<float, 3> f;

System::DelayS(8);

Timer::TimerInt lastTime = System::Time();

while (true)
{
	if (System::Time() - lastTime >= 20)
	{
		myCar.doBlink();

		myCar.myAccel.Update();

		f = myCar.myAccel.GetAngle();

		for (Byte i = 0; i < 3; i++)
			myCar.myUart.SendBuffer((Byte *)&f[i], 4);

		lastTime = System::Time();
	}
}
```

where...
myCar.myAccel Is Mma8451q
myCar.myUart Is UartDevice

---------------------------------------------------------------------------------------------------
I just wonder if it is a memory leak because i found the following description.
```C++
vectors.c  line 432:
	// Read the Fault Address Registers. These may not contain valid values.
	// Check BFARVALID/MMARVALID to see if they are valid values
	// MemManage Fault Address Register
	_MMAR = (*((volatile unsigned long*)(0xE000ED34)));
	// Bus Fault Address Register
	_BFAR = (*((volatile unsigned long*)(0xE000ED38)));
```

nkming2, 2015-01-27:
Actually what I want is a main() that when run will trigger the bug. This is to wipe out other possibilities, e.g., it's your code that bugged. Nevertheless I'll try again and see what happen this time.

nkming2, 2015-01-27:
For your question, i believe it's memory leak.

nkming2, 2015-01-28:
No problem as expected, but I have commented out all unrelated stuff, including the accelerometer. My advice is to check your accelerometer implementation. Closing this for now, but if you have any new info, feel free to reopen the issue.

closed 2015-01-28 by nkming2